FIRST-Tech-Challenge/FtcRobotController #962: Gradle configuration

z3db0y, 2024-03-23:
Hello, I'd like to propose a few changes towards the gradle configuration file:
-  `compileSdkVersion` is upgraded from `29` to `34`
- upgrade gradle (please)

The primary reason for this request is out of the box support for Kotlin, however keeping deps up to date is good practice in general. I could make a PR but I'm too lazy :P

That-Madman, 2024-08-27:
I'd second that (though I'm quite late to the party). Out of the box Kotlin support would be nice, but what would be better is if the gradle config could be changed to not use deprecated features? I had half of my branches start refusing to build when I updated gradle.

z3db0y, 2024-08-30:
Yea gradle is a pain in the ass 90% of the time; you have to align all the versions like stars to get your code to build 😅

ofluffydev, 2024-09-07:
I made a PR with tons of fixes and updates including this one and got it instantly closed by a bot with a message saying "sorry we don't take prs" 😭

texasdiaz, 2024-09-07:
That's because the FtcRobotController repository is the "output" of our internal build process, it's actually automatically generated. We cannot accept PRs to the repo because it cannot go backwards.

If you have suggestions for updates, you can send a link to your PR to ftctech@firstinspires.org and we'll consider it.

-Danny

ofluffydev, 2024-09-27:
Welp fix the build process to not use things that are years behind please?

texasdiaz, 2024-09-27:
> Welp fix the build process to not use things that are years behind please?

Comments like this are not helpful. A lot of things you depend on no longer work properly in newer versions of Gradle, and there are multitudes of build processes that you're not aware of that must be retrofitted because they also break in newer versions. Also, with the planned obsolescence of the FTC SDK build environment there's only so much work we're going to do - the ROI just doesn't make sense.

closed 2024-09-27 by texasdiaz