FIRST-Tech-Challenge/FtcRobotController #965: DCMotorEx getVelocity overflow

Ridleyyy, 2024-04-09:
It appears that the maximum return value of `getVelocity()` is limited due to variable size and as a result, when using systems that result in encoder readings with very high ticks per second, the overflow results in an incorrect value for the velocity of the encoder.

Windwoes, 2024-04-28:
NB: this was previously reported at https://github.com/FIRST-Tech-Challenge/SkyStone/issues/241

Windwoes, 2024-09-14:
NB2: The [OctoQuad FTC Edition](https://www.tindie.com/products/digitalchickenlabs/octoquad-ftc-edition-8ch-encoderpwm-interface/), which is now legal for this season, does not experience this problem.

scoobydrvr, 2025-11-03:
Has this issue been resolved? I see that the previous report has been archived.

scoobydrvr, 2025-11-04:
> It has not been resolved, as it is an issue with the Expansion Hub
> firmware, not the SDK. Given that the last expansion hub firmware update
> was in 2018, I do not expect this issue will be fixed.

Well, that's a bummer... I'd there a workaround such as resetting the
counter we can call every so often to avoid this?

gearsincorg, 2025-11-05:
One solution is to calculate the velocity based on the changing encoder value.  If you want to calculate the velocity each cycle,  just  subtract the previous encoder value (saved in a variable)  from the current encoder value.  Use an ElapsedTimer object to measure the cycle time,  Finally, divide the change-in-encoder-value by the elapsed-time to get velocity in counts-per-second.

This will work well if you have very rapid pulses.

closed 2026-10-03 by Windwoes