FIRST-Tech-Challenge/FtcRobotController #999: REV Controller won't vibrate or have touchpad function

greenbear26, 2024-08-07:
I recently bought the new REV controller that was advertised with rumble functionality and touchpad control, like a PS4 controller would. However, when running the test opmodes for rumble and touchpad on the latest version, it wouldn't work. This is strange because version 9.2 notes say that it added REV controller compatibility. I assume it has a similar type to a PS4 controller behind the scenes, but I am unable to locate the code adding this functionality.

Windwoes, 2024-08-11:
Do you have advanced gamepad features enabled in the settings?

greenbear26, 2024-08-11:
No, what is that?

Windwoes, 2024-08-11:
It's an option in the settings on the Driver Station app.

greenbear26, 2024-08-11:
Ok, after doing that I realized that I hadn't updated the robot controller app and driver station app, so I did that and now it works. Thanks for the help!

closed 2024-08-11 by Windwoes