FIRST-Tech-Challenge/FtcRobotController #1025: maximum supported Android SDK level

ZGStuff, 2024-08-25:
I am wondering if Android 15 beta is supported for the FTC software that runs on-device.
The SDK level for Android 15 is 35.

cmacfarl, 2024-08-26:
It is not.

texasdiaz, 2024-08-26:
The maximum practical supported version of Android is Android 10.

NoahAndrews, 2024-08-26:
That's certainly the last officially supported version, and the version we set as the Target SDK. That doesn't mean newer versions definitely won't work though. I've used the Driver Station with much more recent versions.

ZGStuff, 2024-08-28:
got it.
just making sure so that when android 15 comes out of beta and my Driver Station updates it will still work

texasdiaz, 2024-08-28:
What region are you in, and what device are you using? Just remember there's no guarantee that your version of Android won't stop working, but the hope is that it will not.

ZGStuff, 2024-08-28:
Currently in 

> What region are you in, and what device are you using? Just remember there's no guarantee that your version of Android won't stop working, but the hope is that it will not.
Currently in Central Alberta, Canada, near Edmonton. For personal testing, I'm using a pair of Samsung Galaxy S9 devices, and for actual usage, using a REV Control Hub with one of my Galaxy S9 devices and a REV Expansion Hub for extra ports

ZGStuff, 2024-08-28:
This probably isn't a concern, but in case I ever update my device (REV Control Hub), I need to make sure the app still works.

ZGStuff, 2024-08-28:
That, and my team will be switching to the REV Driver Hub this season.

closed 2024-08-28 by ZGStuff