FIRST-Tech-Challenge/FtcRobotController #1036: Android Studio import fails

Jacoblightning, 2024-09-05:
When I try importing the project into Android Studio, it says I have an incompatible JVM and Gradle Version. Upon updating to A compatible version, I always get the following error:

Multiple build operations failed.
&nbsp;&nbsp;&nbsp;&nbsp;Could not create task ':FtcRobotController:generateDebugRFile'.
&nbsp;&nbsp;&nbsp;&nbsp;Could not create task ':TeamCode:processDebugResources'.
Could not create task ':FtcRobotController:generateDebugRFile'.
Cannot use @TaskAction annotation on method IncrementalTask.taskAction$gradle_core() because interface org.gradle.api.tasks.incremental.IncrementalTaskInputs is not a valid parameter to an action method.

My Host is Arch Linux and everything is fully updated.
[Here](https://pasted.drakeerv.com/SN58dV19) is my full compilation log

Jacoblightning, 2024-09-06:
This is a clean build by the way—straight from [this](https://github.com/FIRST-Tech-Challenge/FtcRobotController/tree/6ce588fc79a65fe25afc97211e12b68608d4b424) tree.

cmacfarl, 2024-09-06:
This smells like an incompatibility between gradle and java, or some mutation thereof.

What are the contents of your `gradle/wrapper/gradle-wrapper.properties` file?
What is your JAVA_HOME set to?

Jacoblightning, 2024-09-06:
@cmacfarl 
My `gradle/wrapper/gradle-wrapper.properties` file is the following:

distributionBase=GRADLE_USER_HOME
distributionPath=wrapper/dists
distributionUrl=https\://services.gradle.org/distributions/gradle-8.9-bin.zip
networkTimeout=10000
validateDistributionUrl=true
zipStoreBase=GRADLE_USER_HOME
zipStorePath=wrapper/dists

My JAVA_HOME is not set (or an empty string).

On opening Android Studio, I get this warning: 

Multiple Gradle daemons might be spawned because the Gradle JDK and JAVA_HOME locations are different. Project 'FtcRobotController' is using the following JDK location when running Gradle: '/usr/lib/jvm/java-21-openjdk' The system environment variable JAVA_HOME is: '/home/jacoblightning3/.local/share/JetBrains/Toolbox/apps/android-studio/jbr' If you dont need to use different paths (or if JAVA_HOME is undefined), you can avoid spawning multiple daemons by setting JAVA_HOME and the JDK location to the same path.

Also, I'm Using the latest nightly version of AS (Ladybug Feature Drop 2024.2.2 Nightly)

Jacoblightning, 2024-09-06:
Also, on first load. I get this warning:

`Undefined java.home on the project gradle/config.properties file when using the gradleJvm #GRADLE_LOCAL_JAVA_HOME macro. To mitigate the issue, this was changed to use the Embedded JDK (JetBrains Runtime 21.0.3).`

and the initial build and sync fails with the following (I have tried both and the latest version and none work):

Your build is currently configured to use incompatible Java 21.0.3 and Gradle 7.4.2. Cannot sync the project.

We recommend upgrading to Gradle version 8.9.

The minimum compatible Gradle version is 8.5.

The maximum compatible Gradle JVM version is 17.

Possible solutions:
 - Upgrade to Gradle 8.9 and re-sync
 - Upgrade to Gradle 8.5 and re-sync

Jacoblightning, 2024-09-06:
Nevermind, I just had to downgrade my Java to Java 8.

closed 2024-09-06 by Jacoblightning

cmacfarl, 2024-09-06:
> Nevermind, I just had to downgrade my Java to Java 8.

That's what I was going to suggest, or at least some java less than or equal to 17.