FIRST-Tech-Challenge/FtcRobotController #1076: Missing null check on ColourBlobLocatorProcessor.Builder imageRegion

bubner, 2024-09-21:
Hi, just reporting a small visual-level bug based on the builder of `ColorBlobLocatorProcessor`.

The region of interest parameter is not checked for being null when `.build()` is called, so instead of throwing a helpful message like `You must set a region of interest!` it throws:
```
java.lang.NullPointerException: Attempt to invoke virtual method 'org.opencv.core.Rect org.firstinspires.ftc.vision.opencv.ImageRegion.asOpenCvRect(int, int)' on a null object reference
   at org.firstinspires.ftc.vision.opencv.ColorBlobLocatorProcessorImpl.init(ColorBlobLocatorProcessorImpl.java:130)
 ```

The colour range and contour mode parameters are already checked when the processor is built to supply more helpful exception messages, so it would be helpful to more novice programmers to receive a more helpful exception if they forget to call `setRoi()`.

Windwoes, 2024-09-21:
Thanks for pointing this out, we'll fix it in the next release

bubner, 2025-02-04:
Fixed in v10.2

closed 2025-02-04 by bubner