FIRST-Tech-Challenge/FtcRobotController #1078: Invalid ImageRegion area exceptions

bubner, 2024-09-21:
Hi, I was using the new `ImageRegion` features alongside a `ColorBlobLocatorProcessor` and found that if an illegal image region were constructed, it would crash the entire application with no useful exceptions. I assume other cryptic exceptions would also be thrown for other misuses of `ImageRegion`.

For example, I accidentally put the region of interest coordinates such that the image size would be zero or negative like this:
```java
ImageRegion.asUnityCenterCoordinates(0.1, 0.1, 0.1, 0.1)
```
which on runtime did not throw any exceptions but instead suddenly crashed the RC app. I found in Logcat these errors only:
```
cv::error()             com.qualcomm.ftcrobotcontroller      E  OpenCV(4.10.0) Error: Assertion failed (!_src.empty()) in cvtColor, file /home/ci/opencv/modules/imgproc/src/color.cpp, line 196
libc                    com.qualcomm.ftcrobotcontroller      A  Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 13727 (robotcontroller), pid 13727 (robotcontroller)
```

Although it's obvious that negative/zero image regions shouldn't work, I believe it would be better for `ImageRegion` to perform some internal checks to throw a helpful exception on improper construction like this, as the current error handling is nearly impossible to interpret especially for something as simple as a missed negative.

Windwoes, 2024-09-21:
Thanks for pointing this out, we'll fix it in the next release

bubner, 2025-02-04:
Fixed in v10.2

closed 2025-02-04 by bubner