FIRST-Tech-Challenge/FtcRobotController #1081: Download Offline Blocks Editor and Download All OpModes return "Internal Error"

wendelj, 2024-09-21:
Updated to 10.1 and connected to the Control Hub over Wifi "Download Offline Blocks Editor" and "Download All OpModes" return "Internal Error" now.

cmacfarl, 2024-09-21:
I tested this on 10.1 and both operations successfully resulted in a downloaded zip file.

Please provide the following to help us root cause your problem.

- A screenshot of the error.
- Any errors in the browser's console log.
- The robot controller logs, which may be obtained by going to the Manage page and clicking on the Download Logs button.

wendelj, 2024-09-21:
[robotControllerLog.txt](https://github.com/user-attachments/files/17085421/robotControllerLog.txt)
[image]
[image]
[image]

cmacfarl, 2024-09-21:
The robot controller logs show a possible problem with a file named, "Posssible new auto.blk".

If you use the REV Hardware Client to downgrade to 10.0 do you still see this problem?

wendelj, 2024-09-21:
Downgraded to 10.0 and the issue persists. Deleted the OpMode named "Posssible new auto" thinking that would resolve the issue but that didn't work either.

cmacfarl, 2024-09-21:
Can you attempt to reproduce again after deleting the "Possible new auto" and send a new set of robot controller logs?

I'd like to see if the exception in the logs related to that blocks program is now gone or not.  Ideally, you'd reproduce and immediately grab the logs so that when searching through them I can focus on a smaller period of time.

wendelj, 2024-09-21:
[robotControllerLog(3).txt](https://github.com/user-attachments/files/17085821/robotControllerLog.3.txt)
Attached log after deleting "Posssible new auto".

texasdiaz, 2024-09-22:
I'm curious if this might be a browser issue. I see you're using Firefox, does this also happen with chrome?

This is likely going to require you making a full backup of your control hub (using the REV Hardware Client) and putting the backup file onto a google drive for us to access. This will allow us to hopefully replicate your problem - this really sounds like some kind of file naming issue that is silently killing the task on the backend (likely an internal file like a config file or something) but the backup will hopefully tell us for certain.

wendelj, 2024-09-22:
I tried Chrome and the Rev Hardware Client as well. Rev Hardware Client opens the file save dialog like it's going to work but it doesn't save anything.
[Control Hub 21536-RC_9-22-2024-11.56.13.864.revbkp.zip](https://github.com/user-attachments/files/17090550/Control.Hub.21536-RC_9-22-2024-11.56.13.864.revbkp.zip)

cmacfarl, 2024-09-22:
I'm somewhat confused?  You said, "Rev Hardware Client opens the file save dialog like it's going to work but it doesn't save anything.", but the zip file you attached is 8mb and appears to contain a dozen or so blocks opmodes.

wendelj, 2024-09-22:
Edited previous comment so it's clearer.

cmacfarl, 2024-09-22:
> Edited previous comment so it's clearer.

Oh that makes sense.  The REV Hardware Client is going to use the same HTTP endpoint as the Program and Manage page for the Download operations.  It's the Control Hub that's returning HTTP 500, so any client will encounter that failure.  The REV Hardware Client should however, not fail silently.

It does appear however that the backup feature of the REV Hardware Client does not use the HTTP route as it succeeded.

cmacfarl, 2024-09-23:
The control hub backup was very helpful.

The problem is that parsing of 'Posssible new auto.blk' is throwing an exception.   And if an exception is thrown it aborts the entire process.

Your file looks like this.  As an aside, I have no idea how you got the file into this state.  I was not able to do so using the browser based Blocks editor alone.

```
<xml xmlns="https://developers.google.com/blockly/xml"/>
<?xml version='1.0' encoding='UTF-8' standalone='yes' ?>
<Extra>
<OpModeMeta flavor="TELEOP" group="" autoTransition="" />
<Enabled value="true" />
</Extra>
```

Apparently the Control Hub parser is not tolerant of the shortcut trailing slash in the xml element.  Modifying the file so that it looks like this results in a successful parse.

```
<xml xmlns="https://developers.google.com/blockly/xml">
</xml>
<?xml version='1.0' encoding='UTF-8' standalone='yes' ?>
<Extra>
<OpModeMeta flavor="TELEOP" group="" autoTransition="" />
<Enabled value="true" />
</Extra>
```

Note I removed the trailing slash in the opening xml element and added the formal end tag for that element.

The only way that I know, at present, how to fix this is to use [adb](https://developer.android.com/tools/adb) to remove the offending file.  The referenced website describes how to obtain it if you don't already have it (it is installed along with Android Studio if you have a laptop with Android Studio already installed).

Using GitBash on Windows with adb, the following sequence of commands will delete the offending Blocks file.

```
adb shell
cd /sdcard/FIRST/blocks
rm Posssible\ new\ auto.blk
rm Posssible\ new\ auto.js
```

Ctrl-c will exit out of adb shell when finished.

texasdiaz, 2024-09-23:
I agree with everything Craig said, but I think the more likely scenario is that perhaps you used an external xml file editor (the edit that caused this issue is a common artifact with visual studio's auto complete feature) and uploaded the file via the Windows File Explorer (or a similar tool that uses the USB file storage protocol). ADB is a handy tool but not necessary.

cmacfarl, 2024-09-26:
A fix wherein a malformed file will prevent these two operations from succeeding, malformed files are skipped/ignored, is scheduled for 10.2.

closed 2024-09-26 by cmacfarl