FIRST-Tech-Challenge/FtcRobotController #1086: Inner classes cannot be ran as OpModes

GraysonTinker, 2024-09-24:
For whatever reason, inner classes of any form cannot be run as an OpMode, even when public and properly annotated. If you try, it will simply fail silently within the Driver Station and put the error "Unable to start OpMode [name]" in the console. This doesn't strike me as particularly intentional behavior given the silent failure and the fact this limitation isn't documented anywhere, but if it is I think the error message should be improved to be more specific (similarly to how it would fail if, say, you annotated a class that didn't extend LinearOpMode).

cmacfarl, 2024-09-24:
Please provide a minimal sample class that you think should run correctly but exhibits this behavior.    A pointer to something on GitHub would be ideal.

cmacfarl, 2024-11-06:
Addressed in 10.1.1.

Now produces a message stating inner classes can not be run as OpModes.

closed 2024-11-06 by cmacfarl