FIRST-Tech-Challenge/FtcRobotController #1098: OpModes do not appear on DS after updating RC app to v10.1

lmcginnisno1, 2024-09-26:
I downloaded the RC app from here, as the REV hardware client says to do to update the RC app when using Android Studio, after loading the new app onto the control hub, I cannot see my Teleop's are Autonomous opmodes on the DS dropdown. I have made sure they are in a public class and have the @Teleop(name="", group="") annotation. If you would like to verify this for yourself or look through my code you can find it here: [](https://github.com/lmcginnisno1/FTC-13917_2025)

cmacfarl, 2024-09-26:
Are you certain that you are deploying the workspace that you reference via the GitHub repo in this Issue description?

Can you provide some screenshots of your Android Studio workspace showing a successful deploy from this workspace?

If you remove the @Disabled annotation from one of the pre-packaged sample OpModes does that OpMode not appear as well?

lmcginnisno1, 2024-09-26:
[image]
Can see sample opMode but cannot see any of my opModes on DS

cmacfarl, 2024-09-26:
I did a git clone of https://github.com/lmcginnisno1/FTC-13917_2025, opened it in Android Studio, deployed to my Control Hub and I see a lot of OpModes that are presumably yours.

[image]

lmcginnisno1, 2024-09-26:
Do you see a teleop red, those opmodes are for tuning roadrunner not the teleop.

cmacfarl, 2024-09-26:
> Do you see a teleop red, those opmodes are for tuning roadrunner not the teleop. Or any opmodes under teleop?

Yes.

[image]

Ignore the Blocks OpModes that are obviously not yours.

cmacfarl, 2024-09-26:
> those opmodes are for tuning roadrunner not the teleop.

Also FWIW, Auto Test is yours.

```
@Autonomous(name = "Auto Test", group = "Auto Test")
public class AUTO_Test extends Robot_Auto {
     int m_Analysis;
```

lmcginnisno1, 2024-09-26:
Weird, when I push the code, the only opMode I see is enable/disable dashboard.

lmcginnisno1, 2024-09-26:
Also, when I downgrade to version 10(which has no issues) It changes the hub name to FTC-MYpa and I have to rescan for the expansion hub.

cmacfarl, 2024-09-26:
If you are deploying software through Android Studio, you should not be using the REV Hardware Client to manage versions.   

See https://ftc-docs.firstinspires.org/en/latest/programming_resources/tutorial_specific/android_studio/fork_and_clone_github_repository/Fork-and-Clone-From-GitHub.html#updating-your-fork-and-local-clone for a detailed writeup of how to update a fork of FtcRobotController.

Do you have any changes in your workspace that are not in the repository on GitHub?

Clone your GitHub repo into a fresh, new directory, load that workspace into Android Studio, and with no other changes, deploy that to your Control Hub.

lmcginnisno1, 2024-09-26:
I will try this tomorrow afternoon, I am aware that I don't use the hardware client for RC versions, but I would need it to connect wirelessly to the robot and update the hub OS and firmware.

lmcginnisno1, 2024-09-26:
Thanks for the help, when I cloned the repo into a new project it works now, not sure what the root cause is but it works now.

closed 2024-09-26 by cmacfarl