FIRST-Tech-Challenge/FtcRobotController #1155: New Variables Declared Locally to Functions

derekrprice, 2024-10-12:
When I create a new variable and add it to my Initialize_Variables function, it is declared as local to the function instead of as a class variable.  How do I fix this?

I started with a program from last year which appears to have its variables declared and initialized properly.  It is only new variables that have this problem.

[image]

texasdiaz, 2024-10-12:
The Java output is only a suggestion of how it could be implemented in Java, that Java output is not used at all. All variables in Blockly are global scope, Blockly has no scoping mechanism.

When defining variables that should be scoped to a function, I tend to name the variable with the function name and then the variable name, making it clear when those variables should be used since I will often want the same variable name for different variables in different functions.

For example, my drivetrain function takes in STRAFE, SPEED, and ROTATE variables. My input parameter variables for my function would then be called:
drive_SPEED, drive_STRAFE, and drive_ROTATE.

lizlooney, 2024-10-12:
If you use the variable in another function, the variable will stop being local and the Java code will be updated.

closed 2024-11-06 by cmacfarl