FIRST-Tech-Challenge/FtcRobotController #1174: incorrect gradle classpath version in build.gradle

daviied, 2024-10-16:
in the build.gradle file, it uses gradle classpath 7.2.0 which will cause an error in new Android studio installs. I did create a pull request, however, go fix this change this line:

classpath 'com.android.tools.build:gradle:7.2.0'

to this:

classpath 'com.android.tools.build:gradle:7.4.0'

Thanks!

cmacfarl, 2024-10-16:
If by new Android Studio you mean Ladybug, the recommended fix is to revert to the Java 17 JDK.

Also discussed here: https://ftc-community.firstinspires.org/t/unable-to-connect-to-control-hub-in-android-studio-ladybug/1132/3

And yeah, I realize Java JDK is redundant :-)  I don't know whether it's a JetBrains or Google thing, but labeling that the Gradle JDK, to me, makes me think it's a Gradle thing and not a Java thing when it's actually a Java thing.  The actual solution is to get Java 17, which is completely independent of Gradle.  Gradle depends upon Java, Java does not depend upon Gradle.  Ok.  Enough terminology ranting.

daviied, 2024-10-16:
I agree that downgrading the JDK works, however, changing that to 7.4.0 would just fix it. I have been running it like that for a couple weeks in intella j and I have not noticed really any issues

qwertychouskie, 2024-10-17:
I figured out the full set of things needed to fix building with newer Android Studio versions:

https://github.com/FTCTeam10298/2024-25-code/commit/d870f4ea24de41aeb453d1a219c1b8a8c219790e

While there is still work to do to move away from deprecated features/functionality, this gets things building again and generally functional.  Ideally Gradle could be upgraded all the way to 8.9 and `com.android.tools.build:gradle` to 8.7.1, but I have not yet tested that, as I wanted a more conservative upgrade to start with to get the FTC SDK building again for people as soon as possible.

(Based on https://github.com/FTCTeam10298/2023-24-code/commit/4ed572c77f2049ff71419e3d6324c92214842c1f, thanks @stupenrose for figuring out a bunch of this last season)

AlecHub, 2024-10-19:
> If by new Android Studio you mean Ladybug, the recommended fix is to revert to the Java 17 JDK...

I hate ladybugs, although farmers love them because they are an effective non-chemical pesticide.  In any event, why would anyone have "bug" in the name their software product.  That seems @lizlooney to me.

cmacfarl, 2024-11-06:
Addressed in 10.1.1

closed 2024-11-06 by cmacfarl