FIRST-Tech-Challenge/FtcRobotController #1205: Can't push code on MacBook

TristanToscas, 2024-10-23:
Whenever I try to push code wirelessly I get two errors with the new 10.1 Controller. This is the error, Unknown host 'repo.maven.apache.org: nodename nor servname provided, or not known'. You may need to adjust the proxy settings in Gradle. I also get this error, Task :FtcRobotController:compileDebugLibraryResources FAILED. I need help trying to figure this out

Froze-N-Milk, 2024-10-24:
The first error can be resolved by disconnecting from the robot, connecting to some normal wifi (school wifis can block things, so if you're connected to school wifi and it fails, try your home wifi, or a hotspot) and hitting the hammer button. There's a dependency or two that needs to be downloaded on the first build, that doesn't get downloaded when you sync.

It sounds like the second issue is caused by the change in default Gradle JDK that upgrading to the newest version of Android Studio causes, or some other similar Gradle issue. Check out the two articles in the Gradle section of [cookbook](https://cookbook.dairy.foundation) on how to ensure that you haven't changed the wrong Gradle setting, and that your Gradle JDK is the correct version. Once you've made changes to gradle settings, I recommend restarting Android Studio, then building your project again.

cmacfarl, 2024-11-06:
Android Studio Ladybug Gradle issues are addressed in 10.1.1.

If you still have problems, please reopen a new issue and provide complete build logs.