FIRST-Tech-Challenge/FtcRobotController #1211: Update Project to follow Android build best practices

KyleChamberlin, 2024-10-24:
Convert gradle build scripts from groovy to kotlinscript for better IDE integration and code completion

Migrate raw gradle `includes` statements into a build plugin.

update Gradle to 8.10+

upgrade AGP to 8.7.1+

move dependency declaration to a version catalog
(`gradle/libs.version.toml`) (using the default location)

create a dependency bundle for the ftc dependencies ('libs.bundles.ftc')

KyleChamberlin, 2024-10-24:
FWIW, I already did this in a clean branch and can issue a PR for that, but seems like something that would make using the repo much simpler.

Windwoes, 2024-10-25:
We internally working on how to update our tooling and evaluating the side-effects of such an update to teams who may be using older versions of Android Studio without an easy update path (e.g. school teams). Updating the tooling is not merely an exercise in bumping version numbers; this repository is merely an autogenerated output of an internal repository with a much more complicated build process, which uses Gradle features that completely break the build in newer versions.

tylerbenson, 2024-11-04:
@Windwoes please consider adding comments to the relevant places in the build documenting why this older version is kept.  I unknowingly went through the update process myself and was about to submit a PR when I saw this.

cmacfarl, 2024-11-06:
The Ladybug problems are addressed in 10.1.1.

As to the rest, the repository you see here is built from a private development repository.  Both need to build and the public repository depends heavily on how the private repository is being built.  Changing how this is all done is not on FIRST's roadmap at present.

closed 2024-11-06 by cmacfarl