FIRST-Tech-Challenge/FtcRobotController #1218: Sdk not working (Grade Error)

NikhilKonduru8, 2024-10-27:
"Deprecated Gradle features were used in this build, making it incompatible with Gradle 9.0.

You can use '--warning-mode all' to show the individual deprecation warnings and determine if they come from your own scripts or plugins.

For more on this, please refer to https://docs.gradle.org/8.9/userguide/command_line_interface.html#sec:command_line_warnings in the Gradle documentation.

BUILD FAILED in 2s"

"Multiple build operations failed.
    Could not create task ':FtcRobotController:generateDebugRFile'.
    Could not create task ':TeamCode:processDebugResources'.
Could not create task ':FtcRobotController:generateDebugRFile'.
Cannot use @TaskAction annotation on method IncrementalTask.taskAction$gradle_core() because interface org.gradle.api.tasks.incremental.IncrementalTaskInputs is not a valid parameter to an action method."

these are the errors that I am getting.

Froze-N-Milk, 2024-10-27:
This issue is caused by the new version of Android Studio automatically upgrading the Gradle JDK version that it uses. Take a look at the [Gradle section of cookbook](https://cookbook.dairy.foundation/gradle/downgrading_gradle_jdk/downgrading_gradle_jdk.html), it contains instructions on how to resolve the issue.

texasdiaz, 2024-10-27:
Also please view this android studio installation note on ftc-docs:
https://ftc-docs.firstinspires.org/en/latest/programming_resources/tutorial_specific/android_studio/installing_android_studio/Installing-Android-Studio.html#configuring-android-studio-ladybug-and-later

AlecHub, 2024-10-28:
The repo's [README](../#Requirements) says you need "_**Android Studio 2021.2 (codename Chipmunk) or later**_":

>## Requirements
> To use this Android Studio project, you will need Android Studio 2021.2 (codename Chipmunk) or later ...

Most teams will not read past "**_or later_**" so perhaps the cautions concerning Ladybug should be noted in a more prominent location.

alan412, 2024-10-28:
> The repo's README <http://../#Requirements> says you need "*Android

You'll notice that the last update to this repo was prior to Ladybug coming
out.   We are working on a solution that will work with Ladybug.

Thanks for your patience!

AlecHub, 2024-10-29:
> You'll notice that the last update to this repo was prior to Ladybug coming out...

I'm so out of touch; I'm still running Hedgehog:

[image]

cmacfarl, 2024-11-06:
Addressed in 10.1.1