FIRST-Tech-Challenge/FtcRobotController #1232: Event Loop Manager throwing IndexOutOfBoundsException

js802025, 2024-10-31:
Every so often the robot will Emergency Stop with an IndexOutOfBoundsException

Here is an excerpt from the log file with the error:

```10-30 17:47:29.354   972  1190 E EventLoopManager: Event loop threw an exception
10-30 17:47:29.355   972  1168 V Robocol : sending CMD_GAMEPAD_LED_EFFECT(10755), attempt: 0
10-30 17:47:29.359   972  1190 E EventLoopManager: java.lang.IndexOutOfBoundsException: Index: 9, Size: 9
10-30 17:47:29.362   972  1190 E EventLoopManager: 	at java.util.ArrayList.get(ArrayList.java:411)
10-30 17:47:29.368   972  1190 E EventLoopManager: 	at org.firstinspires.ftc.robotcore.internal.opmode.TelemetryImpl$LogImpl.get(TelemetryImpl.java:521)
10-30 17:47:29.373   972  1190 E EventLoopManager: 	at org.firstinspires.ftc.robotcore.internal.opmode.TelemetryImpl.saveToTransmitter(TelemetryImpl.java:787)
10-30 17:47:29.378   972  1190 E EventLoopManager: 	at org.firstinspires.ftc.robotcore.internal.opmode.TelemetryImpl.tryUpdate(TelemetryImpl.java:722)
10-30 17:47:29.383   972  1190 E EventLoopManager: 	at org.firstinspires.ftc.robotcore.internal.opmode.TelemetryImpl.tryUpdateIfDirty(TelemetryImpl.java:692)
10-30 17:47:29.388   972  1190 E EventLoopManager: 	at com.qualcomm.robotcore.eventloop.opmode.LinearOpMode.internalOnEventLoopIteration(LinearOpMode.java:223)
10-30 17:47:29.394   972  1190 E EventLoopManager: 	at com.qualcomm.robotcore.eventloop.opmode.OpModeManagerImpl.checkOnActiveOpMode(OpModeManagerImpl.java:801)
10-30 17:47:29.398   972  1190 E EventLoopManager: 	at com.qualcomm.robotcore.eventloop.opmode.OpModeManagerImpl.runActiveOpMode(OpModeManagerImpl.java:444)
10-30 17:47:29.400   972  1190 E EventLoopManager: 	at com.qualcomm.ftccommon.FtcEventLoop.loop(FtcEventLoop.java:288)
10-30 17:47:29.402   972  1190 E EventLoopManager: 	at com.qualcomm.robotcore.eventloop.EventLoopManager$EventLoopRunnable$1.run(EventLoopManager.java:324)
10-30 17:47:29.404   972  1190 E EventLoopManager: 	at com.qualcomm.robotcore.util.ThreadPool.logThreadLifeCycle(ThreadPool.java:737)
10-30 17:47:29.405   972  1190 E EventLoopManager: 	at com.qualcomm.robotcore.eventloop.EventLoopManager$EventLoopRunnable.run(EventLoopManager.java:267)
10-30 17:47:29.406   972  1190 E EventLoopManager: 	at com.qualcomm.robotcore.eventloop.EventLoopManager$1.run(EventLoopManager.java:606)
10-30 17:47:29.412   972  1190 E EventLoopManager: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
10-30 17:47:29.414   972  1190 E EventLoopManager: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
10-30 17:47:29.415   972  1190 E EventLoopManager: 	at com.qualcomm.robotcore.util.ThreadPool$ThreadFactoryImpl$1.run(ThreadPool.java:793)
10-30 17:47:29.415   972  1190 E EventLoopManager: 	at java.lang.Thread.run(Thread.java:761)
10-30 17:47:29.416   972  1190 V EventLoopManager: RobotCoreException in EventLoopManager
10-30 17:47:29.420   972  1190 E EventLoopManager: com.qualcomm.robotcore.exception.RobotCoreException: EventLoop Exception in loop(): IndexOutOfBoundsException - Index: 9, Size: 9
10-30 17:47:29.421   972  1190 E EventLoopManager: 	at com.qualcomm.robotcore.eventloop.EventLoopManager$EventLoopRunnable$1.run(EventLoopManager.java:333)
10-30 17:47:29.422   972  1190 E EventLoopManager: 	at com.qualcomm.robotcore.util.ThreadPool.logThreadLifeCycle(ThreadPool.java:737)
10-30 17:47:29.423   972  1190 E EventLoopManager: 	at com.qualcomm.robotcore.eventloop.EventLoopManager$EventLoopRunnable.run(EventLoopManager.java:267)
10-30 17:47:29.424   972  1190 E EventLoopManager: 	at com.qualcomm.robotcore.eventloop.EventLoopManager$1.run(EventLoopManager.java:606)
10-30 17:47:29.424   972  1190 E EventLoopManager: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
10-30 17:47:29.425   972  1190 E EventLoopManager: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
10-30 17:47:29.426   972  1190 E EventLoopManager: 	at com.qualcomm.robotcore.util.ThreadPool$ThreadFactoryImpl$1.run(ThreadPool.java:793)
10-30 17:47:29.427   972  1190 E EventLoopManager: 	at java.lang.Thread.run(Thread.java:761)
10-30 17:47:29.427   972  1190 V RobotCore: EventLoopManager state is EMERGENCY_STOP
10-30 17:47:29.427   972  1190 V RobotCore: Robot Status: EMERGENCY STOP
10-30 17:47:29.430   972  1164 D RobotCore: system telemetry: key=$System$Error$ msg="System exception: IndexOutOfBoundsException - Index: 9, Size: 9"
10-30 17:47:29.432   972  1190 I FtcEventLoop: ======= TEARDOWN =======```

alan412, 2024-10-31:
What version of the robot controller are you using?

Also, can you please send the code of the OpMode you are running?

Thanks,

Alan

js802025, 2024-10-31:
We're using version 10.1.0.
Here is the link to the OpMode:
https://github.com/fwprobotics/3507-IntoTheDeep/blob/master/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/teleop/TeleOpZones.java