FIRST-Tech-Challenge/FtcRobotController #1344: Deprecating current DcMotor run mode naming

Iris-TheRainbow, 2024-12-02:
Currently, the SDK has three run modes to expose the Lynx firmware functionality for motors, those being `RUN_WITHOUT_ENCODER`, `RUN_USING_ENCODER`, and `RUN_TO_POSITION` for the standard set power mode, the velocity PID, and the positional PID. 

As someone who spends a lot of time helping inexperienced programmers learn, I can pretty confidentially say that few new programmers properly interpret what the run modes actually do from their names. Many, many people assume (in my opinion, quite reasonably) that you must set the motor mode to `RUN_USING_ENCODER` to use `.getPosition()`. In reality, `RUN_USING_ENCODER` is a velocity control mode, which can lead to very confusing behavior when used without understanding what that mode actually does.

I propose that in a future SDK release, `RUN_USING_ENCODER` and `RUN_WITHOUT_ENCODER` should be deprecated and replaced by `RUN_TO_VELOCITY` and `SET_POWER` (or similar, like `RUN_SET_POWER` for consistency) to clear up the actual function of the run mode.

AlecHub, 2024-12-02:
A `RUN_AT_CONSTANT_SPEED` run mode was requested a long long time ago.  Also, a new SDK method was requested to set the desired speed of the motor, such as`DcMotor.setSpeed(int encoderTicksPerSecond)`.

Froze-N-Milk, 2024-12-02:
as far as I'm aware, DcMotorEx.setVelocity() exits, which does what you say

Iris-TheRainbow, 2024-12-02:
Yea that naming is also more accurate imo. Speed is non directional,
velocity is and .setvelocity() can be positive or negative

AlecHub, 2024-12-02:
Thank you @Froze-N-Milk !!!

garrettsummerfi3ld, 2024-12-03:
I have personally ran into those issues in 2016 when I was actively competing. Seeing as this hasn't changed is quite surprising as this was _**unbelievably**_ easy to get wrong and lose sanity over.