FIRST-Tech-Challenge/FtcRobotController #1355: When creating commits, please add more details than just version numbers and a large amount of insertions/deletions with little to no explanation.

ofluffydev, 2024-12-06:
Commit messages that only include a version number, or "Merge x with x" without any descriptive context are highly problematic, particularly in repositories intended for forking and independent use. Such vague messages leave users guessing about what changes were made, complicating updates and making it harder to assess the impact of the changes on their modified versions. This can lead to wasted time, unnecessary conflicts, and a lack of trust in the repository’s maintainability. Descriptive commit messages are crucial for transparency, effective communication, and smoother workflows for all users. 

An example commit I've had this issue with:
https://github.com/FIRST-Tech-Challenge/FtcRobotController/commit/9f11128c611cda46f92c0cbb4cb929322aa1ad70

The last legitimate commit title was all the way back in 2020. You don't necessarily need a lot of details in your title, but by all means please include them in your descriptions.

While it would be nice if the changes made during November, including the one linked, were updated to fix this, you don't have to solve this. 

At the very least, from here on out, help us all out by including:

1. Summary of Changes: A brief description of what was modified, added, or fixed.
2. Reason for the Changes: Explain why the changes were necessary.
3. Impact of Changes: Highlight how the changes affect functionality, compatibility, or performance.
4. Relevant Details: Include any specifics like affected files, functions, or features.
5. References: Mention related issues, pull requests, or tickets (if applicable).

While this minimal style of commit messages may work well for projects within a closed team where everything is discussed outside of GitHub, it significantly hinders those using the repository independently. Users who rely on the commit history to understand updates are left in the dark, making it harder to incorporate changes and maintain their forks effectively. Please address this issue to support the broader community using this project.

Please do not disregard or close this issue without addressing the concern, as this is a recurring problem that likely affects many users/teams.

ofluffydev, 2024-12-06:
A simple response of agreement or instructions on how we are expected to incorporate changes that aren't described anywhere will be more than enough to close this issue.

closed 2024-12-06 by texasdiaz

cmacfarl, 2024-12-06:
@ofluffydev The release process for the FtcRobotController workspace is somewhat unique and does not model what you would expect in a normal git/GitHub repository.

The only commits you will see here are related to packaged releases of the robot controller software.  95% of the changed software in any given release is in the artifacts published to Maven.  As such, the commits that were made in the development repository and which were rolled into the new libraries are not reflected here.   What you see in your referenced commit is entirely gradle changes to support the uprev gradle version, reference the 10.1.1 artifacts.

The README functions as the release notes and documents what actually changed of relevance to teams.  That is where you should first go to see what has changed.   Additionally javadocs are hosted at https://javadoc.io/doc/org.firstinspires.ftc.

This model is similar to other large software projects which release out of private repositories/workspaces.