FIRST-Tech-Challenge/FtcRobotController #1371: why i cant found setmotortest in offine block code ( i following the creating op mode at ftc resource libary

bayunquan, 2024-12-18:
and may i ask the sample program of this new season ftc is it the one that on resource libary ? https://ftc-docs.firstinspires.org/en/latest/programming_resources/tutorial_specific/blocks/creating_op_modes/Writing-an-Op-Mode-with-FTC-Blocks.html

bayunquan, 2024-12-18:
the set motor test problem is fix( so why can i find the sample program of the new season ftc

texasdiaz, 2024-12-18:
Sample Opmodes can be found when you create a new OpMode - you will see a list of them in the Sample drop-down, as shown below:

[image]

Choose a sample and see what it does. If you have questions, feel free to ask.