FIRST-Tech-Challenge/FtcRobotController #1428: UVC Webcam functional on Control Hub USB 2.0 port but fails to initialize on 3.0 port

ftc19743, 2025-01-30:
We have a UVC webcam (ArduCam B0454) we have been using successfully this season on our Control Hub 2.0 port.  We would like to move it over to the USB 3.0 port.  When we do, the webcam is recognized on the USB bus and can be configured, but the FTC libraries fail to bring it online.  Given that USB 3.0 is theoretically backward compatible with USB 2.0 devices, this is surprising.  Below is what we see in the log when we try this.  Emphasis added on a few lines.  

We are keen to make this change to avoid the documented issues with the wifi chip and USB 2.0 port sharing a power supply, leading to disconnects.  We had understood that it was also possible to connect a UVC webcam to the control hub USB C port, but cams connected that way are not recognized on the bus during configuration.

We would greatly appreciate any insights on how we might make this work.

Thank you,

2025-01-30 14:48:28.145   983-1527  RobotCore               com.qualcomm.ftcrobotcontroller      I  Attempting to switch to OpMode Examine Cam
2025-01-30 14:48:28.152   983-1527  RobotCore               com.qualcomm.ftcrobotcontroller      I  ******************** START - OPMODE Examine Cam ********************
2025-01-30 14:48:28.341   983-1527  HardwareMap             com.qualcomm.ftcrobotcontroller      D  Clearing which device instances have been retrieved
2025-01-30 14:48:28.344   983-1527  ThreadPool              com.qualcomm.ftcrobotcontroller      V  container(0x0720b30d: OpModeExecutor) added id=240 TID=0 count=1
2025-01-30 14:48:28.346   983-1535  RobotCore               com.qualcomm.ftcrobotcontroller      V  thread: 'OpModeThread' starting...
2025-01-30 14:48:28.348   983-1535  RevBlinkinLedDriver     com.qualcomm.ftcrobotcontroller      V  Pattern: BLACK, 99, 0.747500
2025-01-30 14:48:28.354   983-1535  System.out              com.qualcomm.ftcrobotcontroller      I  OpenCvCameraBase ctor: EasyOpenCV version 1.7.3
2025-01-30 14:48:28.364   983-1181  Robocol                 com.qualcomm.ftcrobotcontroller      V  sending CMD_NOTIFY_INIT_OP_MODE(11732), attempt: 0
2025-01-30 14:48:28.365   983-1181  Robocol                 com.qualcomm.ftcrobotcontroller      V  sending CMD_STREAM_CHANGE(11733), attempt: 0
2025-01-30 14:48:28.368   983-1536  CameraManager           com.qualcomm.ftcrobotcontroller      D  doOpenCamera()...
2025-01-30 14:48:28.369   983-1536  CameraManager           com.qualcomm.ftcrobotcontroller      D  requesting permission for camera: Webcam(YL20230518V0)
2025-01-30 14:48:28.382   983-1536  WebcamNameImpl          com.qualcomm.ftcrobotcontroller      D  requesting permission for /dev/bus/usb/001/004
2025-01-30 14:48:28.382   983-1536  WebcamNameImpl          com.qualcomm.ftcrobotcontroller      V  asyncRequestUsbPermission()...
2025-01-30 14:48:28.387   983-1536  WebcamNameImpl          com.qualcomm.ftcrobotcontroller      D  permission already available for /dev/bus/usb/001/004
2025-01-30 14:48:28.389   983-1536  ThreadPool              com.qualcomm.ftcrobotcontroller      V  container(0x0eecb4b2: default threadpool) added id=242 TID=0 count=1
2025-01-30 14:48:28.389   983-1536  WebcamNameImpl          com.qualcomm.ftcrobotcontroller      V  ...asyncRequestUsbPermission()
2025-01-30 14:48:28.390   983-1536  Continuati...nchronizer com.qualcomm.ftcrobotcontroller      D  awaiting(camera open)...
2025-01-30 14:48:28.391   983-1537  WebcamNameImpl          com.qualcomm.ftcrobotcontroller      D  permission for /dev/bus/usb/001/004=true
2025-01-30 14:48:28.391   983-1537  CameraManager           com.qualcomm.ftcrobotcontroller      D  permission granted for camera: Webcam(YL20230518V0)
2025-01-30 14:48:28.392   983-1537  CameraManager           com.qualcomm.ftcrobotcontroller      D  asyncOpenCamera(Webcam(YL20230518V0))...
**2025-01-30 14:48:28.393   983-1537  CameraManager           com.qualcomm.ftcrobotcontroller      D  asyncOpenWebcam(Webcam(YL20230518V0))...**
2025-01-30 14:48:28.393   983-1537  UsbResiliantWebcam      com.qualcomm.ftcrobotcontroller      V  construct(0x0d0f193c)
2025-01-30 14:48:28.394   983-1537  UsbResiliantWebcam      com.qualcomm.ftcrobotcontroller      D  openAssumingPermission()...
**2025-01-30 14:48:28.401   983-1537  UvcContext              com.qualcomm.ftcrobotcontroller      D  found webcam: usbPath=/dev/bus/usb/001/004 vid=0x1BCF pid=0x284C serial=YL20230518V0 product=Arducam 5MP Camera Module**
2025-01-30 14:48:28.401   983-1537  Uvc                     com.qualcomm.ftcrobotcontroller      D  [jni_context.cpp:120] Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcContext_nativeCreateUvcDevice()...
2025-01-30 14:48:28.401   983-1537  Uvc                     com.qualcomm.ftcrobotcontroller      V  API: Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcContext_nativeCreateUvcDevice: enter...
2025-01-30 14:48:28.402   983-1537  libusb                  com.qualcomm.ftcrobotcontroller      D  [800.162842] [00000601] libusb: debug [op_create] op_create: szUsbPath=/dev/bus/usb/001/004 szSysfsDir=1-1
2025-01-30 14:48:28.402   983-1537  libusb                  com.qualcomm.ftcrobotcontroller      D  [800.163085] [00000601] libusb: debug [initialize_device] initialize_device(szUsbPath=/dev/bus/usb/001/004 szSysfsDir=1-1)
2025-01-30 14:48:28.403   983-1537  Uvc                     com.qualcomm.ftcrobotcontroller      D  [device.cpp:613] uvc_create_uvc_device()...
2025-01-30 14:48:28.403   983-1537  Uvc                     com.qualcomm.ftcrobotcontroller      D  [libuvc/libuvc_internal.h:347] uvc_device::uvc_device()...
2025-01-30 14:48:28.403   983-1537  Uvc                     com.qualcomm.ftcrobotcontroller      D  [libuvc/libuvc_internal.h:376] acquired vid=7119 & pid=10316
2025-01-30 14:48:28.403   983-1537  Uvc                     com.qualcomm.ftcrobotcontroller      D  [libuvc/libuvc_internal.h:347] ...uvc_device::uvc_device()
2025-01-30 14:48:28.404   983-1537  Uvc                     com.qualcomm.ftcrobotcontroller      D  [device.cpp:613] ...uvc_create_uvc_device() rc=0
2025-01-30 14:48:28.404   983-1537  Uvc                     com.qualcomm.ftcrobotcontroller      V  API: Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcContext_nativeCreateUvcDevice: ...exit
2025-01-30 14:48:28.404   983-1537  Uvc                     com.qualcomm.ftcrobotcontroller      D  [jni_context.cpp:120] ...Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcContext_nativeCreateUvcDevice()
2025-01-30 14:48:28.404   983-1537  UvcDevice               com.qualcomm.ftcrobotcontroller      V  construct(pointer=0x7e69fb48c0|null)
2025-01-30 14:48:28.405   983-1537  Uvc                     com.qualcomm.ftcrobotcontroller      D  [jni_device.cpp:108] Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcDevice_nativeGetLibUsbDevice()...
2025-01-30 14:48:28.405   983-1537  Uvc                     com.qualcomm.ftcrobotcontroller      D  [jni_device.cpp:108] ...Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcDevice_nativeGetLibUsbDevice()
2025-01-30 14:48:28.405   983-1537  LibUsbDevice            com.qualcomm.ftcrobotcontroller      V  construct(pointer=0x7e672a1340)
**2025-01-30 14:48:28.406   983-1537  UvcContext              com.qualcomm.ftcrobotcontroller      D  usb device is *not* UVC compatible, /dev/bus/usb/003/003**
2025-01-30 14:48:28.410   983-1537  UvcDevice               com.qualcomm.ftcrobotcontroller      D  open()...
2025-01-30 14:48:28.495   983-1537  Uvc                     com.qualcomm.ftcrobotcontroller      D  [jni_device.cpp:50] Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcDevice_nativeSetUsbDeviceInfo()...
2025-01-30 14:48:28.495   983-1537  Uvc                     com.qualcomm.ftcrobotcontroller      D  [jni_device.cpp:50] ...Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcDevice_nativeSetUsbDeviceInfo()
2025-01-30 14:48:28.495   983-1537  Uvc                     com.qualcomm.ftcrobotcontroller      D  [jni_device.cpp:233] Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcDevice_nativeOpenDeviceHandle()...
2025-01-30 14:48:28.495   983-1537  Uvc                     com.qualcomm.ftcrobotcontroller      V  API: Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcDevice_nativeOpenDeviceHandle: enter...
[... cut ...]
2025-01-30 14:48:35.082   983-1537  UvcDevice               com.qualcomm.ftcrobotcontroller      D  claimInterface(0) succeeded
2025-01-30 14:48:35.082   983-1537  Uvc                     com.qualcomm.ftcrobotcontroller      D  [libuvc/libuvc_internal.h:784] claimInterface(0) succeeded
2025-01-30 14:48:35.082   983-1537  Uvc                     com.qualcomm.ftcrobotcontroller      D  [libuvc/libuvc_internal.h:771] ...uvc_device_handle::claimInterface() rc=0
2025-01-30 14:48:35.082   983-1537  Uvc                     com.qualcomm.ftcrobotcontroller      D  [device.cpp:279] ...uvc_open() rc=0

ftc19743, 2025-01-30:
Of possible interest:  If we put a powered USB Hub (like the Rev UltraUSB Hub) in between the Control Hub 3.0 port and the webcam, the failure mode shifts to this:

2025-01-30 15:11:28.657   449-520   EventHub                system_server                        I  New device: id=4, fd=89, path='/dev/input/event2', name='Arducam 5MP Camera Module', classes=0x80000401, configuration='', keyLayout='/system/usr/keylayout/Generic.kl', keyCharacterMap='/system/usr/keychars/Generic.kcm', builtinKeyboard=false, wakeMechanism=EPOLLWAKEUP, usingClockIoctl=true
2025-01-30 15:11:28.657   449-520   InputReader             system_server                        I  Device added: id=4, name='Arducam 5MP Camera Module', sources=0x00002103
**2025-01-30 15:11:33.145   983-1310  Continuati...nchronizer com.qualcomm.ftcrobotcontroller      E  deadline expired during await()**
2025-01-30 15:11:33.145   983-1310  Continuati...nchronizer com.qualcomm.ftcrobotcontroller      D  ...awaiting(camera open)
2025-01-30 15:11:33.145   983-1310  CameraManager           com.qualcomm.ftcrobotcontroller      D  ...doOpenCamera()
**2025-01-30 15:11:33.145   983-1310  VisionPortalImpl        com.qualcomm.ftcrobotcontroller      E  Camera opening failed.**

Windwoes, 2025-01-31:
```
2025-01-30 14:48:28.495   983-1537  Uvc                     com.qualcomm.ftcrobotcontroller      D  [device.cpp:279] uvc_open()...
2025-01-30 14:48:28.495   983-1537  Uvc                     com.qualcomm.ftcrobotcontroller      D  [libuvc/libuvc_internal.h:357] uvc_device::open()...
2025-01-30 14:48:28.495   983-1537  libusb                  com.qualcomm.ftcrobotcontroller      D  [800.255892] [00000601] libusb: debug [op_open] op_open: szUsbPath=/dev/bus/usb/001/004
2025-01-30 14:48:28.495   983-1537  Uvc                     com.qualcomm.ftcrobotcontroller      D  [libuvc/libuvc_internal.h:357] ...uvc_device::open() rc=0
2025-01-30 14:48:28.495   983-1537  Uvc                     com.qualcomm.ftcrobotcontroller      D  [device.cpp:295] calling libusb_reset_device()...
2025-01-30 14:48:28.504   425-519   EventHub                system_server                        I  Removing device Arducam 5MP Camera Module due to epoll hang-up event.
```

I believe the smoking gun is here. Some webcams are not tolerant of USB device resets, and it looks like this is one of them. Skipping the USB device reset was one of the workarounds needed to support the Arducam OV9281 in SDK v8.2.

If you're feeling adventurous, you could try adding your camera's VID and PID to the `CamCompatFeature::AvoidLibUsbResetDevice` case in [libuvc_internal.h](https://github.com/OpenFTC/Extracted-RC/blob/master/RobotCore/src/main/cpp/libuvc/include/libuvc/libuvc_internal.h#L307)

ftc19743, 2025-01-31:
Thank you!  We are definitely adventurous!  However, modifying the SDK at the C++ level and rebuilding the libraries and getting those rebuilt libraries incorporated in our Android Studio build would be a new journey for us.  Are there some documents around describing how to do this in general with the SDK?  We think we understand the change to the code you are proposing, but not so clear on how to edit/rebuild that library?
Thanks!

Windwoes, 2025-01-31:
The link I sent points to Extracted-RC, which is already set up to allow you build the SDK entirely from library sources. You should just be able to clone the repo and build in AS like any other project.

ftc19743, 2025-01-31:
Do you happen to know what versions of Gradle and the Gradle JVM should be used to build this project in AS LadyBug?  We have tried various combinations of gradle 8.5/8.9 and Gradle JVM 21/17 and no luck so far.  Haven't gotten past gradle sync.

Here is the gradle sync error with 8.5 and 17:
Could not create task ':Hardware:generateDebugRFile'.
Cannot use @TaskAction annotation on method IncrementalTask.taskAction$gradle_core() because interface org.gradle.api.tasks.incremental.IncrementalTaskInputs is not a valid parameter to an action method.

Windwoes, 2025-01-31:
I haven't tried to build it from Ladybug, you might need to do a side by side install of an earlier version of AS.

ftc19743, 2025-01-31:
Backed up to last release of AS Koala and it built no problem with our tweak to libuvc_internal.h.  That was with jbr-17 and gradle 7.2  Thanks!

To incorporate this into our build, is it as simple as copying the generated "~/FTCRobotController/build/outputs/aar/FtcRobotController-debug.aar" over to the corresponding directory in our team AS project?  

Or perhaps the entire contents of the "~/FTCRobotController/build" directory?  Or even the entire "~/FTCRobotController/" directory?

Windwoes, 2025-01-31:
Let's start with just trying to run a sample vision OpMode from Extracted-RC and see if it fixes the issue.

ftc19743, 2025-01-31:
Good point.  Tried a controlled experiment:

- Reverted any changes to libuvc_internal.h in Extracted-RC
- Enabled ConceptAprilTag op mode
- Plugged a known-functional logitech into the 2.0 port
- Build and install

Sadly it died during initialization.  Seems like something is amiss in the Extracted-RC build we did (fork/clone/build in AS).
Here is the log:
2025-01-31 13:47:12.001  1610-1720  Robocol                 com.qualcomm.ftcrobotcontroller      V  received command: CMD_SET_MATCH_NUMBER(15653) 0
2025-01-31 13:47:12.003  1610-1720  Robocol                 com.qualcomm.ftcrobotcontroller      V  received command: CMD_INIT_OP_MODE(15654) Concept: AprilTag
2025-01-31 13:47:12.023  1610-1860  ThreadPool              com.qualcomm.ftcrobotcontroller      V  container(0x0946bf06: OpModeStuckCodeMonitor) added id=144 TID=0 count=1
2025-01-31 13:47:12.027  1610-1861  RobotCore               com.qualcomm.ftcrobotcontroller      V  thread: ...terminating 'OpModeThread'
2025-01-31 13:47:12.030  1610-1860  RobotCore               com.qualcomm.ftcrobotcontroller      I  Attempting to switch to OpMode Concept: AprilTag
2025-01-31 13:47:12.030  1610-1861  ThreadPool              com.qualcomm.ftcrobotcontroller      V  container(0x0ccee225: OpModeExecutor) removed id=143 TID=1861 count=1
2025-01-31 13:47:12.037  1610-1860  RobotCore               com.qualcomm.ftcrobotcontroller      I  ******************** START - OPMODE Concept: AprilTag ********************
2025-01-31 13:47:12.105  1610-1860  HardwareMap             com.qualcomm.ftcrobotcontroller      D  Clearing which device instances have been retrieved
2025-01-31 13:47:12.107  1610-1860  ThreadPool              com.qualcomm.ftcrobotcontroller      V  container(0x0ad5e1c6: OpModeExecutor) added id=145 TID=0 count=1
2025-01-31 13:47:12.108  1610-1876  RobotCore               com.qualcomm.ftcrobotcontroller      V  thread: 'OpModeThread' starting...
2025-01-31 13:47:12.111  1610-1724  Robocol                 com.qualcomm.ftcrobotcontroller      V  sending CMD_NOTIFY_INIT_OP_MODE(5795), attempt: 0
2025-01-31 13:47:12.112  1610-1876  AprilTagDetectorJNI     com.qualcomm.ftcrobotcontroller      D  Creating 36h11 tag family
2025-01-31 13:47:12.112  1610-1876  AprilTagDetectorJNI     com.qualcomm.ftcrobotcontroller      D  Initializing april tag detector
2025-01-31 13:47:12.341  1610-1876  System.out              com.qualcomm.ftcrobotcontroller      I  OpenCvCameraBase ctor: EasyOpenCV version 1.7.1
2025-01-31 13:47:12.343  1610-1877  CameraManager           com.qualcomm.ftcrobotcontroller      D  doOpenCamera()...
2025-01-31 13:47:12.344  1610-1877  CameraManager           com.qualcomm.ftcrobotcontroller      D  requesting permission for camera: Webcam(989ECEC0)
2025-01-31 13:47:12.351  1610-1724  Robocol                 com.qualcomm.ftcrobotcontroller      V  sending CMD_STREAM_CHANGE(5800), attempt: 0
2025-01-31 13:47:12.353  1610-1877  WebcamNameImpl          com.qualcomm.ftcrobotcontroller      D  requesting permission for /dev/bus/usb/003/005
2025-01-31 13:47:12.353  1610-1877  WebcamNameImpl          com.qualcomm.ftcrobotcontroller      V  asyncRequestUsbPermission()...
2025-01-31 13:47:12.356  1610-1877  WebcamNameImpl          com.qualcomm.ftcrobotcontroller      D  permission already available for /dev/bus/usb/003/005
2025-01-31 13:47:12.356  1610-1877  ThreadPool              com.qualcomm.ftcrobotcontroller      V  container(0x0ab94c9e: default threadpool) added id=147 TID=0 count=1
2025-01-31 13:47:12.356  1610-1877  WebcamNameImpl          com.qualcomm.ftcrobotcontroller      V  ...asyncRequestUsbPermission()
2025-01-31 13:47:12.357  1610-1878  WebcamNameImpl          com.qualcomm.ftcrobotcontroller      D  permission for /dev/bus/usb/003/005=true
2025-01-31 13:47:12.357  1610-1877  Continuati...nchronizer com.qualcomm.ftcrobotcontroller      D  awaiting(camera open)...
2025-01-31 13:47:12.358  1610-1878  CameraManager           com.qualcomm.ftcrobotcontroller      D  permission granted for camera: Webcam(989ECEC0)
2025-01-31 13:47:12.358  1610-1878  CameraManager           com.qualcomm.ftcrobotcontroller      D  asyncOpenCamera(Webcam(989ECEC0))...
2025-01-31 13:47:12.359  1610-1878  CameraManager           com.qualcomm.ftcrobotcontroller      D  asyncOpenWebcam(Webcam(989ECEC0))...
2025-01-31 13:47:12.359  1610-1878  UsbResiliantWebcam      com.qualcomm.ftcrobotcontroller      V  construct(0x0d0bfdd9)
2025-01-31 13:47:12.359  1610-1878  UsbResiliantWebcam      com.qualcomm.ftcrobotcontroller      D  openAssumingPermission()...
2025-01-31 13:47:12.364  1610-1878  UvcContext              com.qualcomm.ftcrobotcontroller      D  found webcam: usbPath=/dev/bus/usb/003/005 vid=0x46D pid=0x825 serial=989ECEC0 product=null
2025-01-31 13:47:12.364  1610-1878  Uvc                     com.qualcomm.ftcrobotcontroller      D  [jni_context.cpp:120] Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcContext_nativeCreateUvcDevice()...
2025-01-31 13:47:12.364  1610-1878  Uvc                     com.qualcomm.ftcrobotcontroller      V  API: Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcContext_nativeCreateUvcDevice: enter...
2025-01-31 13:47:12.365  1610-1878  libusb                  com.qualcomm.ftcrobotcontroller      D  [409.526088] [00000756] libusb: debug [op_create] op_create: szUsbPath=/dev/bus/usb/003/005 szSysfsDir=3-1.4
[... cut ...]
2025-01-31 13:47:13.268  1610-1881  RobotCore               com.qualcomm.ftcrobotcontroller      I  exiting match logcat for /storage/emulated/0/FIRST/matchlogs/Match-0-Concept:_AprilTag.txt
2025-01-31 13:47:15.625   801-801   lvjinhua                android.process.media                D   Failed while trying resolve alarm filter file, not exists
2025-01-31 13:47:17.734  1610-1621  art                     com.qualcomm.ftcrobotcontroller      I  Background partial concurrent mark sweep GC freed 152894(5MB) AllocSpace objects, 77(5MB) LOS objects, 40% free, 16MB/27MB, paused 2.678ms total 116.288ms

Windwoes, 2025-01-31:
Ooops, looks like I forgot to bump some dependency versions in the gradle script for Extracted-RC when it was updated to 10.0.

Can you edit [this dependency block](https://github.com/OpenFTC/Extracted-RC/blob/master/Vision/build.gradle#L35) to have the following versions?

```
    api 'org.openftc:easyopencv:1.7.3'
    api 'org.openftc:apriltag:2.1.0'
```

ftc19743, 2025-02-01:
- After updating those depenencies, Extracted-RC built and ran ConceptAprilTag successfully with a known-to-function Logitech webcam.
- Then switched to the Arducam on the 2.0 port and ConceptAprilTag also ran successfully.
- Then modified the AvoidLibUsbResetDevice case in hasFeature() to always return true (for now) and the opmode ran successfully again.  
- Notably: "**[device.cpp:291] libusb_reset_device() not called**" was printed in the log.
- Moved the Arducam over to the 3.0 port and the opMode.......... .............................................................................................Worked!

You truly a team saver Windwoes!  Much appreciation for your help.

We could use a couple more hints: 

(1) We were not quite able to decipher the way the VID and PID are being used in libuvc.  We see the 16 bit shifts, but we are a bit confused because the existing Arducam you had support for only had a 3 byte VID (0xC45) while most of the others have 4 bytes.  We can report that the log reports this for our Arducam:  
 "found webcam: usbPath=/dev/bus/usb/001/002 vid=0x1BCF pid=0x284C serial=YL20230518V0 product=Arducam 5MP Camera Module". 
Can you give us a hint on how to structure the AvoidLibUsbResetDevice case properly?

(2) We think the way to integrate this change into our robot build is to add a "libs" folder to our project and move the generated "~\Extracted-RC\RobotCore\build\outputs\aar\RobotCore-debug.aar" file from Extracted-RC to that new libs folder.  Then we need to add this to the repositories block in build.dependencies.gradle:
flatDir {
        dirs 'libs'
    }
And then replace this line in the implementation block
    implementation 'org.firstinspires.ftc:RobotCore:10.1.0'
With
    implementation(name: 'RobotCore-debug', ext: 'aar') 
Is that even remotely correct?

Thank you SO much!

Windwoes, 2025-02-01:
> (1) We were not quite able to decipher the way the VID and PID are being used in libuvc. We see the 16 bit shifts, but we are a bit confused because the existing Arducam you had support for only had [...]
"found webcam: usbPath=/dev/bus/usb/001/002 vid=0x1BCF pid=0x284C serial=YL20230518V0 product=Arducam 5MP Camera Module".
Can you give us a hint on how to structure the AvoidLibUsbResetDevice case properly?

The VID is still 4 bytes, but the leading 0 is just implied. `0x0C45` is the same thing as `0xC45`, just like `0.5` is equivalent to `.5`. That being said it should probably be changed in the code to be `0x0C45` so that it's consistent with the others.

For structuring the case you'd just want to add your VID and PID to `usbconstants.h` and then add case like:

```c
 case makeVidPid(VENDOR_ID_ARDUCAM2, PRODUCT_ID_ARDUCAM_B0454):
```

> (2) We think the way to integrate this change into our robot build is to ... Is that even remotely correct?

Yeah that's roughly the right idea, but you might have issues with the other libraries that have a transitive dependency on RobotCore. You might have to add a gradle exclude block to those to block them from still trying to pull RobotCore from maven.

However it may just be easier to copy-paste your TeamCode source files into Extracted-RC and use that as your primary repo.

The other question though is one of legality. In the past, the game manual expressly prohibited modifying the portions of the SDK which are packages as binary `.aar` files, which is exactly what we are doing here. This year's game manual appears to be silent on this topic, so I'm not sure if it would be an issue or not.

ftc19743, 2025-02-02:
Thanks for the heads up.

We are so close.  We modified the uvc code as you suggested and tested and its all good.  Now just a matter of incorporating that tweak into our build.

We tried adding the updated robotcore aar to our build but couldn't even get gradle to find it on a sync.  We'll keep working on that.

We also tried incorporating our code (and a couple of additional library dependencies like Acme Dashboard) into our fork of Extracted-RC but that build failed with a cascade of duplicate java symbols like this:
   > Duplicate class com.google.gson.DefaultDateTypeAdapter found in modules jetified-RobotCore-8.2.0-runtime (org.firstinspires.ftc:RobotCore:8.2.0) and jetified-gson-2.8.0 (com.google.code.gson:gson:2.8.0)
Thoughts on that?

Windwoes, 2025-02-02:
> We also tried incorporating our code (and a couple of additional library dependencies like Acme Dashboard) into our fork of Extracted-RC but that build failed with a cascade of duplicate java symbols like this:

This smells very strongly of the library author incorrectly using `implementation` or `api` for the RobotCore dependency, rather than `compileOnly`.

Windwoes, 2025-02-03:
Note the following on the FTC Dash readme:

[image]

ftc19743, 2025-02-04:
Ahh! That was probably the issue with our attempt to build inside of Extracted-RC.  Thank you!

In the meantime, we thought we would try something really clever.  So we set up a local maven repository on our build computer, used maven to "install" the updated version of the RobotCore aar file there with the same maven coordinates as the original version, then added a line to the repositories section _above_ all the others that pointed to our local repository.

To our delight it built, with one small issue where, for some reason, the transitive dependency from RobotCore to threetenabp didn't register.  Adding that dependency to our build.dependencies.gradle resulted in what we thought was a successful build.  However, when we installed it on the ControlHub, things went south quickly.  The RobotController app couldn't successfully initialize and we had to do a factory reset to get it back...as you said up front, quite the adventure!

We'll try it the way you suggested (from inside Extracted-RC) again with the exclusion you pointed out above.  THX!

ftc19743, 2025-02-04:
Some more good news!  Those exclusions did the trick.  Here is our modified build.gradle (TeamCode module) that pulls in the two external libraries we use (both built on the FTC SDK):

```
repositories {
    mavenCentral()
    google() // Needed for androidx
    maven { url = 'https://maven.brott.dev/' }
    maven { url 'https://s01.oss.sonatype.org/content/repositories/snapshots/'}
}

dependencies {
    /*
     * Other Modules
     */
    implementation project(':FtcRobotController')
    implementation project(':Hardware')
    implementation project(':RobotCore')
    implementation project(':FtcCommon')
    implementation project(':Vision')
    implementation('com.acmerobotics.dashboard:dashboard:0.4.15') {
        exclude group: 'org.firstinspires.ftc'
    }
    implementation('com.w8wjb.ftc:ftc-neodriver:1.0-SNAPSHOT') {
        exclude group: 'org.firstinspires.ftc'
    }
    implementation 'org.firstinspires.ftc:gameAssets-FreightFrenzy:1.0.0'
}
```

The only other tweak needed was a bump in the minSdkVersion from 23 to 24 since our NeoDriver library declared a minSDKVersion of 24.  We made that change in build.common.gradle.

We've built the RobotController with our code and robot config files on several machines, successfully installed it on the Control Hub, ran our test code for calibrating our computer vision (**with the arduCam plugged into the USB 3.0 port**), and all seems well.  If there are other issues with the build, perhaps they will manifest as we run through the rest of our testing.

We think this completes our adventure for now.  Thank you again for helping us out with this, it was very educational and will put our minds at ease regarding wifi disconnects.

If support for the ArduCam B0454 were to make it into the next minor release of the SDK, or even a team configuration of some sort to disable the USB reset on the webcam (since that seems to be a common issue), that would be really great.  We think the B0454 could be an interesting solution for many teams.  It's relatively inexpensive ([$32 on Amazon](https://www.amazon.com/gp/product/B0972KK7BC/)), has a much wider field of view than anything comparable, and has a very small form factor that can be incorporated into many builds.  It led us to learn quite a bit about lens and perspective distortion and how those can be fixed in OpenCV as well.

THANK YOU!

Windwoes, 2025-02-04:
Happy to help