FIRST-Tech-Challenge/FtcRobotController #1448: Problem on Android Studio, repo.maven.apache.org

SamS34, 2025-02-08:
I have a problem on Android Studio, this is for ftc robot the problem is I can connect to ADB over WiFi, but after that when I build to  TeamCode, It starts building but than I get an issue of - Gradle Sync issues

Unknown host 'No such host is known ([repo.maven.apache.org](http://repo.maven.apache.org/))'. You may need to adjust the proxy settings in Gradle.

Enable Gradle 'offline mode' and sync project
[Learn about configuring HTTP proxies in Gradle](https://docs.gradle.org/current/userguide/userguide_single.html#sec:accessing_the_web_via_a_proxy)

It does not let me do anything because

 of that and this is all happening after we had updated the robot

If anyone knows how to fix it than please help me 

Thanks

[image]
[image]
[image]

acharraggi, 2025-02-08:
Is this a new install of Android Studio or a new setup of the FtcRobotController project? The 'unknown host' error is likely be due to no internet connection.  Connect to the internet and build the project. That will downloads files as part of the Gradle sync. After that, it should build without an internet connection and you should be able to ADB over wifi.

Froze-N-Milk, 2025-02-09:
> Is this a new install of Android Studio or a new setup of the FtcRobotController project? The 'unknown host' error is likely be due to no internet connection. Connect to the internet and build the [...]

Yep, you need to build. A Gradle Sync is not a complete build, and it misses a couple of dependencies that are required for it to be deployed offline.

SamS34, 2025-02-10:
This all happens when i build, i have a button which I use too connect to my robot, than when I build I get the problem and yes this is Ladybug the new version of Android Studio and all this is happening after my team updated the driver hub

cmacfarl, 2025-02-10:
> This all happens when i build, i have a button which I use too connect to my robot, than when I build I get the problem and yes this is Ladybug the new version of Android Studio and all this is happening after my team updated the driver hub

Disconnect from the robot.  Ensure you are connected to the internet and can see the FTC artifacts on Maven Central by visiting this url in your browser.

  https://central.sonatype.com/search?q=org.firstinspires.ftc

Once you've confirmed that, select Build -> Rebuild Project in Android Studio.

[image]

Gradle sync, while you are connected to the internet should download and cache the artifacts, but if it didn't, the rebuild project should work.

Once you have confirmed a successful build, reconnect to the Control Hub and deploy as normal.

cmacfarl, 2025-02-10:
Assuming this is resolved.  

If it is not, please post a question over on https://ftc-community.firstinspires.org/

closed 2025-02-10 by cmacfarl

SamS34, 2025-02-10:
k thanks

SamS34, 2025-02-15:
I tried to question on the FTC website and I had gotten no response after a week, and when I do the steps that @cmacfarl told me I get stuck on the part where I "rebuild project." 

[image]

than

[image]

Froze-N-Milk, 2025-02-15:
have you tried pressing on the blue link in the error? If not, give it a go, otherwise, search up how to turn off offline mode for gradle in Android Studio. (I distinctly remember the error links that are supposed to turn off offline mode not working, but that may have been fixed). Then, try rebuilding the project again, and send the new error if you get one. best of luck.

SamS34, 2025-02-15:
Ya no it doesn't work I have been trying to mess around with that stuff for a month now and have searched up so much but I have not found anything that can solve this issue. It does not work, thanks for you effort

Samuel Singh

cmacfarl, 2025-02-15:
> I tried to question on the FTC website and I had gotten no response after a week,

I monitor the FTC community forums pretty closely and I haven't seen a post regarding this issue over on https://ftc-community.firstinspires.org/.

Can you please post the url of the question you posted?

I am personally at a loss to describe what you are seeing.  I have not seen this error appear on my system.  It is either a problem with connecting to the internet so you can download the required artifacts from Maven. which is implied by the "Unknown host" message in your first image, or it's a weird setting in Android Studio Ladybug that somehow got set, but honestly, I poked around a bit locally and have no real clue what that might be.  Have you tried opening the Android Studio settings dialog and searching for offline?

Maybe more drastically, uninstall Android Studio, and reinstall from scratch, hopefully to reset all your Android Studio settings?

SamS34, 2025-02-22:
Ok so the part where to restarting Android Studio I have done that 2 times. Also I have got no message from FTC community?

texasdiaz, 2025-02-22:
To my knowledge you have not made any posts to the https://ftc-community.firstinspires.org forum. If this is in error, can you provide a link to your post?

I highly recommend starting from scratch with a new project in a new location on your computer. And is your computer online when it compiles, on an internet connection that has zero blockers (must NOT be on a typical school network).

cmacfarl, 2025-02-22:
> Ok so the part where to restarting Android Studio I have done that 2 times. 

As to restarting Android Studio, I don't think that will solve your problem.  I was suggesting _uninstalling_ and _resinstalling_ Android Studio entirely, and if or when it asks you if you want to retain Android Studio configuration, say, "No."

Then follow @texasdiaz's starting from scratch advise.

> Also I have got no message from FTC community?

I asked you to respond with the URL of the post you made at https://ftc-community.firstinspires.org/.  You have not done so.  You won't get a message if you don't post to the forums.