FIRST-Tech-Challenge/FtcRobotController #1464: Radians normalize issue

edibucura, 2025-02-23:
There is a known bug going around the FTC scene at the moment related to the normalization of radians in the SDK.
This was noticed because the Gobilda Pinpoint driver uses the built-in Pose2D to save the pose of the robot. When ESD happens the bytes that are being read might suffer from interference, sometimes making the heading close to infinite.
At the moment the angle normalize function uses a while loop which substracts 2PI continuously. When a very big number is fed into that function the opmode is stuck in that loop for some seconds.
org.firstinspires.ftc.robotcore.external.navigation.AngleUnit.normalizeRadians is the function with the issue

alan412, 2025-02-24:
Thanks for mentioning this here.   I have logged it as an issue for it to be fixed in the future.  (Probably not until the off-season release).   In the meantime, you can fix it by putting this in your code and calling instead of normalizeRadians
```java
     double myNormalizeRadians(double radians ){
        if(Math.abs(radians) > 4 * Math.PI){
            double numRevolutions = Math.floor(Math.abs(radians) / (2*Math.PI));
            radians = Math.signum(radians) * (Math.abs(radians) - (numRevolutions * (2*Math.PI)));
        }
        return AngleUnit.normalizeRadians(radians);
  }

alan412, 2025-02-24:
In your version of pedropathing, you could put the call to your normalizeRadians here:
https://github.com/Pedro-Pathing/PedroPathing/blob/c1c4aa426f96901ad7c63a13e0455bc1b532866f/src/main/java/com/pedropathing/localization/localizers/PinpointLocalizer.java#L179

You could also modify your copy of the GoBilda pinpoint driver.   (Obviously do one or the other, but not both.)

Hope this helps!!

--Alan

qwertychouskie, 2025-06-28:
Is this fixed in 10.3?  I didn't see it in the changelog.

closed 2025-06-28 by cmacfarl