FIRST-Tech-Challenge/FtcRobotController #1470: `ServoImpl#getPosition` incorrectly reversing a scaled servo.

kevinthegreat1, 2025-03-02:
In `ServoImpl#getPosition`,`reverse` is called on the unscaled position in `getPosition`. While in `ServoImpl#setPosition`, `reverse` is called on scaled values. This means `Servo#getPosition` return incorrect values for reversed scaled servos.

## Example
Let's say we have a reversed servo scaled from `0` to `0.5` instead of from `0` to `1`:

Setting:
1. `setPosition(0.5)` -> sets an actual position of `0.25` for the servo.

Getting:
1. `ServoController#getServoPosition` returns `0.25` (line 178)
2. `reverse` reverses the value from `0.25` to `0.75` (line 179)
3. `Range.scale` scales the value from `0.75` to `1.5` (line 180)
4. `Range.clip` clips the value to `1`

## Result
**We expect to get 0.5, not 1**

## Code Screenshot
[image]

kevinthegreat1, 2025-03-02:
I know you guys don't accept prs, but I'd be happy to submit one if needed.

alan412, 2025-03-02:
Thank you for this catch.   I have put it as an issue on our internal repo.   We'll likely fix it for the off-season release.

alan412, 2025-03-02:
Also @kevinthegreat1 , all of the detail was very helpful!!

alan412, 2025-03-02:
I think the issue is actually in the setting.   and it should be like this:

# Setting:

setPosition(0.5) -> __SHOULD__ set an actual position of 0.75 for the servo.

# Getting:

ServoController#getServoPosition returns 0.75 (line 178)
reverse reverses the value from 0.75 to 0.25 (line 179)
Range.scale scales the value from 0.25 to 0.5 (line 180)
Range.clip does nothing

@kevinthegreat1 - agree?

kevinthegreat1, 2025-03-02:
Thanks for the quick response. I believe both approaches will have the same effect, however, changing `setPosition` would break current servos.

Since right now, `setPosition(0.5)` sets 0.25, but if it's changed to set 0.75, then it will break more current contraptions and people would have to recalibrate their servos. Changing `getPosition` will break less stuff, but both approaches will fix the issue.