FIRST-Tech-Challenge/FtcRobotController #1505: Add toggle functionality to gamepads

abidingabi, 2025-06-25:
SDK version 10.3 adds rising/falling edge detection to button presses for gamepads. 

I really like this feature, as this is something teams commonly want to do, and the [way of doing it otherwise](https://gm0.org/en/latest/docs/software/tutorials/gamepad.html#rising-edge-detector) is a bit inelegant. 

In the same vein, I think it would be useful to add toggle functionality for gamepad; this is also fairly frequently desired by teams, and the [logic is even less elegant](https://gm0.org/en/latest/docs/software/tutorials/gamepad.html#toggles) than rising/falling edge detectors.

Specifically, I would imagine this looks roughly like adding `gamepad1.aToggled()`, `gamepad1.leftBumperToggled()`, etc.

ftcaarya, 2025-06-27:
Hey so I am a bit new to all this, and actually had a question. After reading this PR I looked into the solution for this and at least me, came to the conclusion that you would have to edit the file "GamepadStateChanges.java" and "Gamepad.java" for the solution that I came up with. However they are in a different package they aren't in the SDK (correct me if I am wrong). So how would someone open a PR to change those files? 

This is their package - `package com.qualcomm.robotcore.hardware;`

abidingabi, 2025-06-27:
Hello, the FTC SDK has a closed development process; they do not generally accept outside contributions. I am not affiliated with the tech team (who writes the SDK), but this was intended a suggestion for them to potentially implement, not for outside contributors. I do appreciate the initiative though :)

@ftcaarya

ftcaarya, 2025-06-27:
Ah! I see. That makes sense. I was confused on as to why you opened a Issue for a typo instead of PRing it yourself. Thank you for the clarification!

alan412, 2025-07-16:
We discussed this in the tech team meeting and decided that the opportunity for confusion for what "toggle" meant and what would be the expected value outweighed the benefits of providing this functionality.  For example, should the toggle start out as false or true.   And what should happen if it toggles multiple times between a user checking it?

However, with the addition of the Gamepad edge detection, implementing a toggle is now trivial.   One way to do it would be with the following code in your OpMode:
```java
if (gamepad1.leftBumperWasPressed())){ 
    intakeToggle = !intakeToggle; 
} 
```

Hope this helps!!

Personally as a coach, I would recommend being very careful about toggles.    It can be very easy for a driver to lose track of the state of a  robot during the heat of competition.

closed 2025-07-16 by cmacfarl