FIRST-Tech-Challenge/FtcRobotController #1546: Error with BuildConfig import in FtcRobotController

Striker909-Celestial, 2025-09-06:
In FtcRobotController/src/main/java/org/firstinspires/ftc/robotcontroller/internal/FtcRobotControllerActivity.java on line 81 `import com.qualcomm.ftcrobotcontroller.BuildConfig;` fails. This appears to be because `com.qualcomm.ftcrobotcontroller` is an invalid address. `com.qualcomm.robotcore.BuildConfig` exists, but this BuildConfig does not have a property that is used in the file

cmacfarl, 2025-09-06:
Can you please try Invalidate Caches and Restart?

Striker909-Celestial, 2025-09-06:
The error persists

Windwoes, 2025-09-06:
I believe this error is normal until you try to build for the first time, since this file is autogenerated during build.

Striker909-Celestial, 2025-09-06:
I feel kinda silly because far from my first time doing this, but thanks for the help. Learned something new I suppose.

cmacfarl, 2025-09-06:
Absolutely nothing to feel silly about.   We've all been there.  Heck I think I was there just this week.  Hope you have a great season, I think this will be a fun game to both play and spectate!